arduino/Arduino #10417: [Library Manager] Update ElegantOTA and AsyncElegantOTA Libraries

ayushsharma82, 2020-06-24:
Hi,
It's been quite some time that my libraries are not getting updated in the Arduino library index.

If possible, can you guys let me know why?

Here are the links to both the libraries:

1. https://github.com/ayushsharma82/AsyncElegantOTA
2. https://github.com/ayushsharma82/ElegantOTA

Thanks!

per1234, 2020-06-24:
Hi @ayushsharma82 all the releases of those libraries are available to me in Library Manager:
[image]
[image]

The only exception is AsyncElegantOTA 1.0.1, because the library.properties `version` value wasn't updated:
https://github.com/ayushsharma82/AsyncElegantOTA/blob/v1.0.1/library.properties#L2

closed 2020-06-24 by per1234

ayushsharma82, 2020-06-24:
Yeah! Actually I used VSCode Arduino Extension which didn't update the library index and upon opening the Arduino IDE library manager, the index was updated.

Thanks for your support @per1234 !