arduino/Arduino #10425: Icon change for macOS Big Sur

ardacebi, 2020-06-25:
There's an ongoing conversion to square-rounded cornered icons in macOS with the upcoming update Big Sur. All stock apps have started using these icons and many 3rd party apps are starting with the change as well.

Are there any plans for Arduino IDE?

[image]

baranonen, 2020-06-30:
[image]
I've made something that looks more like Big Sur icons

ardacebi, 2020-07-02:
If I get some response or feedback from one of the assignees or maintainers, I'd love to send out @baranonen's icon on Apple's design board. This solves the shadow and provides the necessary icon sizes for macOS.

facchinm, 2020-07-02:
@sbhklr :arrow_up:

sbhklr, 2020-07-02:
Thanks for reporting @ardacebi and thank you for the proposal @baranonen ! Good timing, I've just read about this today :-) We'll look into this.

ardacebi, 2020-07-02:
Great.

> I'd love to send out @baranonen's icon on Apple's design board. 

Let me know if you're interested in this.

[image]

sbhklr, 2020-07-02:
@ardacebi What is Apple's design board? I don't remember having heard about it. Can you elaborate?

ardacebi, 2020-07-02:
@sbhklr The one I sent above. Didn't know what to call it so... It's a template to generate the icons on the required sizes from Photoshop.

tychoregter, 2020-08-26:
> @sbhklr The one I sent above. Didn't know what to call it so... It's a template to generate the icons on the required sizes from Photoshop.

Do you mind sharing the psd file?

ardacebi, 2020-08-26:
@tychoregter The empty template file provided by Apple or the template file with @baranonen's icon?

guyongithub, 2020-11-15:
Here's my go at an Arduino icon for Big Sur. I made it using Figma and this [template.](https://www.figma.com/community/file/857303226040719059) I'm a noob at graphic design so the execution is far from the best but I thought it's good enough to share anyways.

[image]

If you want to use it:
- Download the image
- Open it in Preview, then select all (Command + A), and then copy (Command C)
- Use finder to find Arduino, select it, go to Get Info (command I)
- Click on the icon and press Command V to paste it.