arduino/Arduino #10426: Feature Request: allow custom linker scripts

JetForMe, 2020-06-25:
Because my project is solar powered, the power to it is sometimes very "dirty." This is causing the bootloader to trigger (despite a bug I fixed). I want to just flash my code without any bootloader, but that means changing the linker script, which currently can only be done platform-wide.

It'd be great if I could include a linker script in my sketch like any other source file.

Thanks!

facchinm, 2020-06-25:
Hi @JetForMe ,
are you using a SAMD board? In that case I'd move the issue on the proper repo (and give you a possible hint :wink: )

JetForMe, 2020-06-25:
I am using SAMD! But I'm sorry, I don't know where the proper repo is. Seems like this would apply to any platform though, wouldn't it?

Thanks!