arduino/Arduino #10430: Add rhio-LIS2HH12 library

JoseMartiRhombio, 2020-06-26:
Hello,

rhomb.io have develop a LIS2HH12 library to run the this accelerometer
Please, could you add the https://gitlab.com/rhombio/rhio-LIS2HH12 to the Library Manager?

Thanks

cmaglie, 2020-06-30:
Done.

JoseMartiRhombio, 2020-07-02:
Excuse me, I have needed to change the url, could you update the information with this url; https://gitlab.com/rhombio/rhio-libraries/rhio-LIS2HH12 ?

JoseMartiRhombio, 2020-07-02:
> Done.

Excuse me, I have needed to change the url, could you update the information with this url; https://gitlab.com/rhombio/rhio-libraries/rhio-LIS2HH12 ?

cmaglie, 2020-07-07:
The "subgroups" feature of gitlab requires us to implement a special handling on our library indexer. I think this is a quite new feature of gitlab since until now we assumed that the path was always in the form: `http://gitlab.com/USERNAME/REPOSITORY`

BTW I see that gitlab redirects from the old URL to the new URL transparently so things should just work in your particular case.

About the missing v0.1.1, the problem is not the changed repo URL but the version written in `library.properties`:

https://gitlab.com/rhombio/rhio-libraries/rhio-LIS2HH12/-/blob/v0.1.1/library.properties

as you can see the file still has `version=0.1.0`. BTW on master I see `0.1.1` so I guess you pushed the tag before updating the version on master. To fix this just delete the tag `0.1.1` and redo it on master`

I'll keep this issue open until we add support for gitlab's "subgroutp" on the library indexer.

kbkibbe, 2020-12-09:
Hello Jose. I've been trying your library on a Sparkfun nRF52832 board. Using your simplest example I am unable to get data from the accel, I just get zeros. The setup of the LIS2HH12 is CS is high and SD0 is low. What I see on a logic analyzer is ONLY a setup write to 0x3A which gets a nak and the a setup read to 0x3B which also gets a NAK. This happens a bunch of times then a ~2 second delay. Any thoughts on where I am going wrong? I'm using ver 0.1.2. Thanks.

JoseMartiRhombio, 2021-01-20:
Hello kbkibbe,

I'm sorry for the delay, I haven't seen your message. You say something about CS but the library doesn't work with SPI bus. If you have this problem yet, you could send me your example code and I'll tell you more about this.

Thanks for contact me