arduino/Arduino #10432: ESP8266 and ESP32 deletes each other group in 1.8.13

skaarj1989, 2020-06-26:
Unable to have ESP8266 and ESP32 groups at the same time in Board Menu.
When I install ESP826 -, ESP32 group is removed (same behavior on the other way).

Bhuvanbhu, 2020-06-26:
I too faced the same issue

matthijskooijman, 2020-06-26:
Do you have links to the board manager URL(s) and/or git repositories for the two board packages you are referring to? Then I can have a closer look at reproducing this.

matthijskooijman, 2020-06-26:
Also, did this work in earlier IDE versions?

Bhuvanbhu, 2020-06-26:
https://arduino.esp8266.com/stable/package_esp8266com_index.json 

https://dl.espressif.com/dl/package_esp32_index.json

skaarj1989, 2020-06-26:
> Also, did this work in earlier IDE versions?

Previous versions didn't have **groups** feature.

matthijskooijman, 2020-06-26:
> Previous versions didn't have groups feature.

They did, but instead of submenus, there would be sections of a single menu (i.e. each installed package would have a heading inside the boards menu, with all of its boards below).

When the submenus were introduced, I think the grouping did not really change, just the way it was displayed (and I quickly checked the code, couldn't find any obvious problems there).

So I'm curious if with previous versions, would the board from both packages be shown in the menu as expected?

Thanks for the URLs, I'll see if I can reproduce.

skaarj1989, 2020-06-26:
Previous versions were fine.
By groups I meant what you called **submenus**

matthijskooijman, 2020-06-26:
Turns out I had both packages already installed, and esp8266 v2.6.3 alongside esp32 v1.0.4 were both shown in the board menu as expected. Upgrading esp8266 to v2.7.1 still works as expected.

 I initially tested with a locally compiled version of the IDE, but switching to an official 1.8.13 version didn't help. I also tried renaming my `~/.arduino15` directory, to start with a clean slate, and just install the ESP8266 and ESP32 packages, but that still works as expected.

[image]

> Just installed 1.8.12, same issue.

Hah, thanks for checking. Then maybe something else changed on your system that coincided with the 1.8.13 upgrade.

Can you tell me:
 - Your operating system and version
 - The version of the ESP8266 and ESP32 cores

Could you maybe see if this can be reproduced with a clean configuration? That would mean renaming your existing configuration (e.g. `C:\Users\YourUsername\AppData\Local\Arduino15`, `~/.arduino15` or `/Users/YourUsername/Library/Arduino15/` depending on your OS), restarting the IDE, re-adding the package.json urls and installing the ESP cores. That will tell us if there is some problem in your Arduino15 directory, or maybe something in the rest of the system.

Afterwards you can rename your original directory back to get your original settings and installed boards back again (but if the problem *also* occurs with the clean settings, maybe do not delete that yet, it might be useful for subsequent testing).

matthijskooijman, 2020-06-26:
Also: You say "When I install ESP826 -, ESP32 group is removed", does that mean that it just does not show the submenu in the boards menu? How does it look in the Boards Manager? Are both marked as "installed" there?

You also say "(same behavior on the other way).", what does that mean exactly? What do you need to do exactly to get ESP32 back and let 8266 be hidden? Uninstall/reinstall ESP32? Change ESP32 version?

skaarj1989, 2020-06-26:
Windows 10 Pro 2004 19041.329

[image]

skaarj1989, 2020-06-26:
Reinstalled 1.6.13, then 1.8.13 ... both packages appeared
[image]

matthijskooijman, 2020-06-26:
Hm, now that's weird... Maybe 1.6.13 fixed something in the Arduino15 directory somehow... You did not rename your Arduino15 directory, I think? Hard to further diagnose this from here. Maybe you could still answer my questions from my previous comment about what triggered this exactly?

Also, if you can again reproduce this, it would be helpful to run arduino from a command prompt (make sure to run `arduino_debug.exe` then) to see if that shows any additional error messages or other output. Also, if it occurs again, I would be interested in a copy of the `C:\Users\YourUsername\AppData\Local\Arduino15` directory (e.g. as a zip), to see if I can reproduce it with that (or can find something otherwise weird).

@Bhuvanbhu, if the problem is still occuring for you, maybe you could try the above (run from command prompt and send over the Arduino15 directory)?

per1234, 2020-08-04:
Since there hasn't been a response in over a month, I'm going to close this. Thousands of people are using both these boards platforms with Arduino IDE 1.8.13 without any further reports of the issue, so it certainly isn't universal.

@Bhuvanbhu and @skaarj1989, if you want to comment here with more information to allow us to investigate further, I'll be happy to re-open this.

A common cause of reports of one 3rd party board platform disappearing after the installation of a new one is that when the user added the new 3rd party Boards Manager URL to their preferences, they remove the previous one.

closed 2020-08-04 by per1234