arduino/Arduino #10453: ARDUINO IDE PORTS ON WINDOWS 10

zeev1, 2020-07-01:
The PORTS options do not appear in Arduino Uno tools when using Windows 10 PC computer. I was expecting to see COM1 or higher. I loaded Arduino IDE on Windows XP computer without an issue. 
[image].
I am using the computer ideacentre AIO 720-241KB

dotto59, 2020-07-03:
First of all, what release of Arduino IDE you're using? I use Arduino IDE 1.8.12 on regular basis under Win 10 without any problem. I hope you didn't install the IDE from Windows Store, as some people had serious problems with that. And are you using the "installer" or "portable" (zip) version?
Anyway it seems to me you have a problem with the USB-Serial device driver. Check the installation (or try to uninstall then re-install it verifying it "dings" when connecting a board, and then the port is visible under Device manager (so I think it'll be visible on you IDE too..).

per1234, 2021-03-29:
Closing due to lack of response.

@zeev1 if you'll provide the requested information I'll be happy to re-open this issue.

closed 2021-03-29 by per1234