arduino/Arduino #10464: Arduino IoT Cloud Dashboard problem with Google-Maps

AndreasWaldherr, 2020-07-03:
Dear Arduino Support Team!

Since two days the Google Map View in my Arduino IoT Dashboard does not work correctly. When I connect to: https://create.arduino.cc/iot/dashboards/622ff3d2-189d-4372-8c08-0be4d76fbcfc - then initialy the Map was displayed as normal, but after a second the Error "Oops! Something went wrong." (in german "Hoppla! Ein Fehler ist aufgetreten.") appears. This happens on all my Browsers and Devices...

Thanks a lot for checking and repairing
Andreas Waldherr

[image]
[image]

facchinm, 2020-07-06:
@mastrolinux @sbhklr   where should this be moved?

sbhklr, 2020-07-06:
@AndreasWaldherr Hi Andreas! We're aware of the issue and currently solving it. Thanks a lot for reporting! If you encounter any other issues please use the bug reporting feature on IoT cloud to get more direct feedback (the red widget on the right side where the scroll bar is). If it's okay for you we would close the issue here.

P.S. Kling ja super spannend dein Bienenstock IoT tracking! Wir arbeiten zurzeit an einem Re-Design the IoT Cloud plattform wie du wahrscheinlich gesehen hast. Damit möglichst viele Nutzer davon profitieren können, versuchen wir die Bedürfnisse der Benutzer besser zu verstehen und diese zu decken. Gerne würden wir mehr darüber erfahren wie du unsere Cloud Plattform nutzt. Hättest du evtl. Zeit und Interesse mit uns bei einem virtuellen Kaffee etwas aus dem Nähkästchen zu plaudern? Das wäre super spannend für uns und du würdest uns dabei helfen die Platform weiterzuverbessern 🙏  Zudem könnten wir dir im Gegenzug ein kleines Goodie aus unserem Store anbieten. :-)

sbhklr, 2020-07-06:
@AndreasWaldherr The map widget issue should be resolved now. Can you confirm?

AndreasWaldherr, 2020-07-06:
Hi Sebastian, 

 

yes I can confirm – it‘s working again 😊

 

Thanks a lot

Andreas

 

PS: Martino, thanks for forwarding…

closed 2020-07-06 by facchinm

sbhklr, 2020-07-06:
@AndreasWaldherr Great! Please let me know in case you're interested in having a chat about your bees and the cloud :-)
If so, you can drop me a line at s.hunkeler@arduino.cc 🙏

AndreasWaldherr, 2020-07-06:
Sebastian, in the future I will use the red feedback widget!

 

Gerne würde ich Euch bei einem virtuellen Kaffee über meine IoT Wünsche und meine Erfahrungen berichten. Ein kleines Geschenk aus dem Store ist wirklich nicht notwendig, denn ich kaufe gerne Eure Produkte um diese großartige Plattform zu stützen und fördern…

 

Ich bin diese Woche von Mittwoch bis Freitag unterwegs, ansonsten habe ich immer Zeit. Sebastian such Dir einen Termin aus, ich richte mich nach dir/euch. 

 

Ganz liebe Grüße

Andreas