arduino/Arduino #10465: Arduino Nano 33 BLE pinout documentation is incorrect

kckemper, 2020-07-04:
At least the mapping for A0 is wrong in both the pinout diagram and the schematic.

- The pinout labels it as A0->AIN[0]
- The schematic labels it as A0->AIN[3]
- The NINA-B3 datasheets list it mapping to A0->AIN[2] (this is correct)

Pleas check over and update the schematic and pinout diagram with the correct part mapping.

sbhklr, 2020-07-06:
@facchinm ☝️ Is that true?

facchinm, 2020-07-06:
I need to take a closer look, no time now...

FunzioneFabio, 2020-07-31:
@agdl Can you doublecheck this please?

agdl, 2020-08-10:
@kckemper thank you for reporting this. In any case the port is right. I would say to remove the AIN[x] label from the pinout since you just use the port value in the end. What do you think?

kckemper, 2020-08-10:
Well, the nrf52840 ADC registers actually only use the AIN pins for configuration. You configure the SAADC with the AIN pin number and it takes over the appropriate port/pin. It's also wrong in the schematic for A0. Some of the other analog pins are mapped ok in the schematic (A7->AIN1 is correct) but is wrong in the pinout document (A7->AIN7 is wrong). I'm fine if you remove it from the pinout document but the schematic is still wrong for at least A0.

FunzioneFabio, 2020-09-17:
@kckemper Thank you for reporting this, the pinouts are now corrected and uploaded on the website.
https://content.arduino.cc/assets/Pinout-NANOble_latest.pdf
https://content.arduino.cc/assets/Pinout-NANOsense_latest.pdf

@kengdahl can you please take care of the schematics?

kengdahl, 2020-09-23:
@FunzioneFabio Sure.

per1234, 2021-03-17:
Hi @kengdahl. Is there any news on this? I see A0 is still associated with AIN3 in the schematics both PDF and Eagle.

kengdahl, 2021-03-22:
@per1234 I have asked HW to make new schematics.