arduino/Arduino #10466: Addition of two libraries to Ardunino

mrfaptastic, 2020-07-04:
Please include two libraries into Arduino's library list:

https://github.com/mrfaptastic/json-streaming-parser2

https://github.com/mrfaptastic/ESP32-RGB64x32MatrixPanel-I2S-DMA

per1234, 2020-07-04:
Hi @mrfaptastic. Your "Json Streaming Parser 2" library looks fine to me, but the "ESP32 LED MATRIX (HUB75) DMA Display" library doesn't currently meet the requirements for addition to the Library Manager index.

From the Arduino library specification:
https://arduino.github.io/arduino-cli/latest/library-specification/#libraryproperties-file-format
>Library names must contain only basic letters (A-Z or a-z) and numbers (0-9), spaces (), underscores (_), dots (.) and dashes (-).

The `name` value "ESP32 LED MATRIX (HUB75) DMA Display" contains disallowed characters `(` and `)`.

If you still want to add your library to the Library Manager index, you'll need to change the name and then either make a new release/tag (making sure to update the library.properties `version` value first), or else redo the existing tag.

After that, please comment here to let us know we can proceed. Thanks!

mrfaptastic, 2020-07-04:
Hi per1234, thanks for the quick review. I have adjusted the library name to remove the brackets.

Kind regards.

per1234, 2020-07-04:
Although the `name` is now compliant in the repository, the Library Manager indexer works from Git tags and your tag still points to the commit from when the `name` value was invalid:
https://github.com/mrfaptastic/ESP32-RGB64x32MatrixPanel-I2S-DMA/blob/1.0.0/library.properties#L1

So you'll need to either create a new tag/release or else redo the existing tag.

mrfaptastic, 2020-07-04:
*facepalm* Sorry. New Tag Created.

cmaglie, 2020-07-07:
Done.

closed 2020-07-07 by cmaglie

njh, 2022-12-18:
@mrfaptastic just seen that [/mrfaptastic/json-streaming-parser2](https://github.com/mrfaptastic/json-streaming-parser2) has disappeared.

Do you have any background into why the repo has gone?

mrfaptastic, 2022-12-20:
I had nuked it, but I've brought it back from the dead now (and improved the library as a bonus).