arduino/Arduino #10470: A few typos and grammar errors in the Arduino IoT Cloud home page

schilli91, 2020-07-04:
I just stumbled upon the following errors in the Android IoT Cloud home page (see https://www.arduino.cc/en/IoT/HomePage).

I would've posted a PR for these changes, but couldn't find the website sources anywhere. Suggested labels for the issue: "Component: Documentation".

### In section "Arduino IoT Cloud Properties"

[...] It might also include a thermostat **_to be able which will change_** the room’s temperature.
**Suggestion:** 
[...] It might also include a thermostat **_which will be able to change_** the room’s temperature.

### In section "Arduino IoT Cloud Events"

The Arduino IoT Cloud becomes aware of events when it receives application messages that indicate **_the_** something has happened. [...]
**Suggestion:**
The Arduino IoT Cloud becomes aware of events when it receives application messages that indicate something has happened. [...]

### In section "Software for Arduino IoT Cloud"

Arduino IoT Cloud allows other methods **_o_** interaction, including HTTP REST API, MQTT, Command-Line Tools, Javascript, and Websockets. [...]
**Suggestion:**
Arduino IoT Cloud allows other methods **_of_** interaction, including HTTP REST API, MQTT, Command-Line Tools, Javascript, and Websockets. [...]

[...] Events can be **_the_** switching the pump on, switching the fan off, turning on the lights, etc.
**Suggestion:**
[...] Events can be switching the pump on, switching the fan off, turning on the lights, etc.

facchinm, 2020-07-06:
@per1234 should this be moved to https://github.com/arduino/reference-en ?

per1234, 2020-07-06:
@facchinm I don't think so, because this specific content is not stored in the arduino/reference-en repo.

I was told that Arduino Create issues should be tracked in the [relevant sections of the Arduino forum](https://forum.arduino.cc/index.php?board=128.0), but I don't know whether that applies to documentation issues.

sbhklr, 2020-07-06:
@per1234 Done. You may close the issue. 
@schilli91 It's gonna take a while until the cache invalidates and the changes become visible. Thanks for your suggestions!