arduino/Arduino #10473: Issue while adding precompile archive

thetechguy2016, 2020-07-05:
Hello everyone

i want to create a library but want to use it with archive file. 

i create a test library and i got some error while compile.

Steps, I followed:
1. Create Library structure 
```
   {Library name (test)}/
                   example/
                           test.ino
                   src/
                       esp8266/
                              libtest.a
                       test.h

```
2. create test.h , test.cpp and test.ino file 

test.h file
 
```
#ifndef TEST_H
#define TEST_H

#if defined(ARDUINO) && ARDUINO >= 100
  #include "Arduino.h"
#else
  #include "WProgram.h"
#endif

#define NAME "AGLOW"
void name();
void data();
#endif
```
test.cpp

```
#include <test.h>
void name() {
    Serial.print(NAME);
}
void data() {
    Serial.print(NAME);
}
```

test.ino file
```
#include <test.h>
void setup() {
    Serial.begin(115200);    
    name();
}
void loop() {}

```

3. I compile the code with `dot_a_linkage=true` then i will get the `test.a` file in temporary folder .

4. now i delete the `test.cpp` file and add `test.a` file inside the esp8266 folder as mentioned above.

5. and change the `test.a` file name to `libtest.a`

6. now i compile the code with this `library.properties`, but nothing works

```
name=TEST
version=1.0.0
author=Vipul Garg 
maintainer=Vipul Garg
sentence=A library use to do
paragraph=Supports HTTPS1.1
category=Communication
url=https://xyz.com
architectures=esp8266
includes=test.h
precompiled=true
ldflag=-ltest
```

7. i got this below error

```
Used: /Users/vipulgarg/Documents/Arduino/libraries/test
/Users/vipulgarg/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /var/folders/4p/d03ndbt94pj3bw26wnzvly580000gn/T/arduino_build_447457/sketch/test1.ino.cpp.o:(.text.setup+0x8): undefined reference to `name()'
/Users/vipulgarg/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /var/folders/4p/d03ndbt94pj3bw26wnzvly580000gn/T/arduino_build_447457/sketch/test1.ino.cpp.o: in function `setup':
/Users/vipulgarg/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/cores/esp8266/HardwareSerial.h:76: undefined reference to `name()'
collect2: error: ld returned 1 exit status

```

8. if i add test.cpp again in the src folder everything works fine.

Please help me out i have tried for more than 5 days still got error.
my test library is attached to this issue.

[test.zip](https://github.com/arduino/Arduino/files/4874860/test.zip)

facchinm, 2020-07-06:
Hi @thetechguy2016 ,
which version of the IDE / arduino-builder are you using? IDE 1.8.13 compiles the test example just fine with the zip you provided.

Please not that IDE 1.8.12 had a very serious bug in precompiled libraries handling and should be updated

thetechguy2016, 2020-07-06:
hi @facchinm 

Thanks for your prior response , earlier i was using Arduino 1.8.10 now i just update it to 1.8.13 but the issue is still the same, can you just check my side if possible to resolve. i am using macos is there any issue with that anyway?

facchinm, 2020-07-06:
Sorry, my fault, I didn't check if the cpp was still in the library.
The problem is that esp8266 core doesn't support precompiled libraries; instead than inserting the ldflags in random places as we used to do, we now require the core creators to provide the hook explicitly (so we don't get funky linking errors).
In verbose mode, the compilation reports
```
Compiling library "test"
Library test has been declared precompiled:
Using precompiled library in /home/martino/eslov-sk/libraries/test/src/esp8266
The plaform does not support 'compiler.libraries.ldflags' for precompiled libraries.
```
Adding the flag in `platform.txt` with this patch solves the issue
```patch
From 527f5c6df2c51b023cfd5dfd0cd9def996b4a805 Mon Sep 17 00:00:00 2001
From: Martino Facchin <m.facchin@arduino.cc>
Date: Mon, 6 Jul 2020 11:10:17 +0200
Subject: [PATCH] Enable precompiled libs

---
 platform.txt | 3 ++-
 1 file changed, 2 insertions(+), 1 deletion(-)

diff --git a/platform.txt b/platform.txt
index 6c0a8910..f7aa7b5c 100644
--- a/platform.txt
+++ b/platform.txt
@@ -80,6 +80,7 @@ compiler.cpp.extra_flags=
 compiler.ar.extra_flags=
 compiler.objcopy.eep.extra_flags=
 compiler.elf2hex.extra_flags=
+compiler.libraries.ldflags=
 
 ## generate file with git version number
 ## needs bash, git, and echo
@@ -102,7 +103,7 @@ recipe.S.o.pattern="{compiler.path}{compiler.c.cmd}" {compiler.cpreprocessor.fla
 recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}"
 
 ## Combine gc-sections, archives, and objects
-recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" {build.exception_flags} -Wl,-Map "-Wl,{build.path}/{build.project_name}.map" {compiler.c.elf.flags} {compiler.c.elf.extra_flags} -o "{build.path}/{build.project_name}.elf" -Wl,--start-group {object_files} "{archive_file_path}" {compiler.c.elf.libs} -Wl,--end-group  "-L{build.path}"
+recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" {build.exception_flags} -Wl,-Map "-Wl,{build.path}/{build.project_name}.map" {compiler.c.elf.flags} {compiler.c.elf.extra_flags} -o "{build.path}/{build.project_name}.elf" -Wl,--start-group {object_files} "{archive_file_path}" {compiler.c.elf.libs} {compiler.libraries.ldflags} -Wl,--end-group  "-L{build.path}"
 
 ## Create eeprom
 recipe.objcopy.eep.pattern=
-- 
2.27.0

``` 
@per1234 I think we already documented this but I don't remember where... Do you have any idea :slightly_smiling_face: ?

thetechguy2016, 2020-07-06:
Hi @facchinm 

It works perfectly now, but i did not find the change in github repo as you commit the change.

And thanks for your support @facchinm.

per1234, 2020-08-01:
@facchinm I don't believe it was previously documented, but I have now added it to the Arduino platform specification: https://github.com/arduino/arduino-cli/pull/840

Since my understanding is that this issue was left open due to the need to make sure this is documented, I'll close it as fixed.

> i did not find the change in github repo as you commit the change.

@thetechguy2016  here it is: https://github.com/arduino/arduino-builder/commit/ffe96696f3447613e7975e7ba99f63db4349ed4c

closed 2020-08-01 by per1234