arduino/Arduino #10485: Cannot access COM10, COM11, etc.

Xigent, 2020-07-09:
Please read the Issues section of the Contributing Rules at the "Contributing" link to the right before submitting an issue report.
The drop down box for COOM Port selection under tools won't let me access above COM9.   My device is at COM11.  Please Help!

per1234, 2020-07-09:
This issue tracker is only to be used to report bugs or feature requests. This topic is more appropriate for the Arduino Forum. I'm sure we'll be able to help you with your problem over there.

Please do this:
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules.

closed 2020-07-09 by per1234

PaulStoffregen, 2020-07-09:
@Xigent - You didn't even tell us which product "My device" really is.  Are you using a genuine Arduino product?  Which one?  Or is a direct clone, or some 3rd party board?  Please consider making a screenshot with both the Windows Device Manager showing COM11 and the Tools > Ports menu, so we can see the same problem you are actually seeing on your screen.  Small details matter.

@per1234 - There is a well known issue that Windows requires special handling for higher than COM9.  As far as I know, the software for all official Arduino boards handle > COM9 properly, which makes me suspect this may in fact be a bug, but likely with some 3rd party software.  Of course it's just a guessing game with so little info given.  If a screenshot is posted and this is confirmed to be an Arduino brand board, I'd recommend reopening this issue.

Xigent, 2020-07-10:
You didn't even tell us which product "My device" really is. Are you using a genuine Arduino product? Which one?
I am using ESP32  NodeMCU-32S.  I included the associated library.  I had it working when it enumerated to COM7.I added additional devices, and it enumerated to COM11 or COM10 depending on boot sequence.I use USBDevView to display all USB configurations and subsequent COM ports.  EVERYTHING is mapped out in this program.  It is a free utility.
But there is no COM10, COM11 in Arduino Program.  I have Two(2) Arduino Uno's also attached to COM Ports and they are working fine.I need to program to access above COM9 if possible, to remedy the plethora of hardware devices hanging on my system.
Can this be done?  Please Help.Thank-you for any help and understanding.

Gregory Keyock(610) 865-5461xigent@verizon.net

PaulStoffregen, 2020-07-10:
This probably is the wrong place to ask for help with ESP32.  Espressif has their own repository and forums.

If this is a software defect (these issue trackers are for reporting software defects & generally not meant for tech support), the problem is almost certainly in the ESP32 portion of the software, since Arduino Uno is working correctly.

I know this doesn't directly help you, but if you truly didn't realize where would be the appropriate & effective place to ask for help, hopefully you can understand that to get help you first need to find the correct place to ask.