arduino/Arduino #10486: Arduino IDE doesn't correctly use the home folder on Linux

Avamander, 2020-07-09:
For some reason the Arduino IDE creates a folder `~/.arduino15` and uses it for both cache and configuration. That's not in-line with [XDG Base Directory Specification](https://wiki.debian.org/XDGBaseDirectorySpecification) and clutters people's home directories.

Arduino IDE should place it's cache in `~/.cache/arduino`, configuration in `~/.config/arduino` and logs in `~/.state/arduino`.

per1234, 2020-08-01:
Closing as duplicate of https://github.com/arduino/Arduino/issues/3915

closed 2020-08-01 by per1234

Avamander, 2020-08-01:
@per1234 that one is closed though. It's still an issue.

per1234, 2020-08-01:
@Avamander it was closed as wontfix. If anyone has unique and productive information to add, they should do so in that issue and the developers can make the decision as to whether it is worth reopening. Fragmenting the discussion across multiple duplicate issue reports is unproductive.

Avamander, 2020-08-01:
@per1234 That decision was made five years ago (by an ex-employee now I think?). It's IMHO whole new discussion.