arduino/Arduino #10489: [library manger] Please add SDI-12 Library

SRGDamia1, 2020-07-09:
Github repo:  https://github.com/EnviroDIY/Arduino-SDI-12

per1234, 2020-07-10:
Hi @SRGDamia1 . Arduino has a policy that 3rd party libraries with a library.properties `name` value starting with the word "Arduino" (or any capitalization thereof) will no longer be accepted for inclusion in the Arduino Library manager index. The reason is that Arduino wants to reserve these names for their own official libraries.

If you want your library to be added to the Library Manager index, you will need to change the `name` value in library properties to one that is compliant with the Arduino Library Specification:
https://arduino.github.io/arduino-cli/library-specification/#libraryproperties-file-format

Then update the `version` value in library.properties and create a new release/tag, or else redo the existing release/tag.

Note that this restriction only applies to the `name` value. You are not required to rename the repository.

Once you have done this, please comment here so we'll know we can proceed.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

SRGDamia1, 2020-07-10:
I'm sorry I missed that requirement.  The name has been changed and a new release issued.

Thank you!

cmaglie, 2020-07-17:
Done.

closed 2020-07-17 by cmaglie