arduino/Arduino #10492: COM problems with Arduino IDE on Virtual machin

Makkio88, 2020-07-10:
Hi, i have a win-10 virtual machine running on vmware, that im using to run arduino ide.
I never  got problems when i use Arduino uno board.
Now i bought a new arduino mkr1010 wifi and i start to experience serious problems.
When i try to compile or when i open serial port monitor, the ide fails to identify correct port causing various kind of problems.
How can i solve? I need to continue using my vm to develope.

per1234, 2020-08-01:
>When i try to compile
...
> the ide fails to identify correct port causing various kind of problems.

The port is not used during compilation, only upload and Serial Monitor. Please provide a detailed description of the problems you're having related to the port while compiling.

>when i open serial port monitor

Do you have the correct port selected in the Arduino IDE's **Tools > Port** menu before opening Serial Monitor? The serial port of the MKR WiFi 1010 works differently than the Uno because the Uno has a dedicated USB to serial adapter chip that provides the port, whereas the MKR WiFi 1010 uses CDC serial on the primary ATSAMD21G18 microcontroller connected directly to your computer. During upload, the MKR WiFi 1010 goes through this process:

1. Upload process starts.
2. The Arduino IDE signals to the sketch program running on the ATSAMD21G18 to reset.
3. ATSAMD21G18 receives the signal and resets itself.
4. The bootloader program runs.
5. The bootloader program creates a CDC serial port.
6. The upload occurs via the bootloader's CDC serial port.
7. After the upload finishes, the bootloader program exits and the sketch program runs.
8. The sketch program creates a CDC serial port.

The fact that this kills and creates a USB device twice is problematic on a VM where you typically have to specifically enable access to USB devices. It's possible you're just experiencing this behavior. It's also possible you're experiencing something completely different. It's impossible to say based on the vague description of the problem you provided.

per1234, 2020-09-25:
Closing due to lack of response.

@Makkio88 if you'll provide the requested information I'll be happy to re-open this issue.

closed 2020-09-25 by per1234