arduino/Arduino #10496: GCC complie time options

trlafleur, 2020-07-12:
The IDE, using the variant.h file knows what CPU the user has selected and then can make a decision on the proper flags to invoke to build the project...

On the compile line is uses the "-mcpu" to define the CPU family to use. "-mcpu=cortex-m4".
This information should ALSO be added as a  "-D"  flag, to give a consistent format for the user to determine the processor class he is using at run time.  -Dcortex-m4

Also as the IDE knows the exact processor it using from the variant.h file, it should also be added as a -D flag. 
-D__SAMD51G19A__ 

In reviewing a number of variant.h file, it's very inconsistent from manufacturer to manufacturer on what -D flags are included.

Doing this would add a very consistent format that code developers could depend on...

matthijskooijman, 2020-07-13:
On AVR, I believe gcc defines such a value based on the `-mmcu=...` option. I think for ARM gcc, this does not happen, so indeed Arduino/core/variants/something needs to handle that externally.

> In reviewing a number of variant.h file, it's very inconsistent from manufacturer to manufacturer on what -D flags are included.

Do you mean variants *within* the same Arduino core? Or from core to core?

I guess that such defines are by definition not portable between architectures (processor families), and thus are essentially a core-specific API (i.e. from the core to the sketch). OTOH, when there are multiple cores for the same processor families (e.g. the Arduino and Adafruit SAMD cores), it would be useful if these defines would be consistent.

This might be something to add a recommendation for in the core documentation (part of `arduino-cli`). If so, this issue might be better moved to the `arduino-cli` repo?

trlafleur, 2020-07-13:
I stopped using 8bit processors a few years back, so my search of the variant.h files were ADAfruit, Arduino, STMicro, ESP32, and Teensy 32 bit, processors...

I also cross-posted this on Arduino-Pro Github site