arduino/Arduino #10500: Mipi camera record video

PaulderGraph, 2020-07-13:
I would like to record a mipi video, wich is 4k resolution. How can I record it? Which microcontroller do I need?

per1234, 2020-07-13:
This issue tracker is only to be used to report bugs or feature requests. This topic is more appropriate for the Arduino Forum. You'll be sure to get an answer to your question over there.

Please do this:
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules.

closed 2020-07-13 by per1234

PaulStoffregen, 2020-07-13:
I do not believe any microcontroller on the market today is up to this task.  You probably need a single board computer like nVidia Jetson.  Google search can help you find those products and their forums to ask for help.