arduino/Arduino #10504: Please add WebSockets2_Generic Library

khoih-prog, 2020-07-14:
Please add [WebSockets2_Generic](https://github.com/khoih-prog/WebSockets2_Generic) to Library Manager. Thanks.

This [WebSockets2_Generic library](https://github.com/khoih-prog/WebSockets2_Generic) is based on and modified from [Gil Maimon's ArduinoWebSockets Library](https://github.com/gilmaimon/ArduinoWebsockets) to provide support to many more boards, such as ***Arduino SAMD21, Adafruit SAMD21/SAMD51, nRF52, etc.*** and enable those boards to use WebSockets services, including voice-control Alexa along with Blynk. 

Those supported boards can also run ***WebSockets Server.*** The WebSockets can be used with ***ESP’s WiFi, WiFiNINA***. Support to STM32, Teensy, SAM DUE and ESP8266/ESP32-AT, W5x00, ENC28J60 and LAN8742A Ethernet will be added in future versions.

khoih-prog, 2020-07-16:
Just update to new v1.0.1

#### New in v1.0.1

1. Add support to ***SAM DUE*** boards.
2. Add support to ***Ethernet W5x00, using either Ethernet, EthernetLarge or Ethernet2 library.***
3. Restructure code to provide flexibility to support many more WiFi/Ethernet modules/shields in the future. Please delete the *.cpp files, replaced by *.hpp files, in the src directory, if *.cpp files still exist after installing new version.

cmaglie, 2020-07-17:
Done.

closed 2020-07-17 by cmaglie