arduino/Arduino #10508: Catalina 10.15.5 does not recognize Arduino Serial ports

paris007, 2020-07-14:
I am running Catalina 10.15.5  on Mac book Pro  (16-inch, 2019).  The serial port does not show up on Arduino IDE and so I am having serious trouble uploading my code on an Itsybitsy nRF52840 Express board.

UnexpectedMaker, 2020-07-20:
> I am running Catalina 10.15.5 on Mac book Pro (16-inch, 2019). The serial port does not show up on Arduino IDE and so I am having serious trouble uploading my code on an Itsybitsy nRF52840 Express board.

@paris007 This is a macOS issue with USB2.0 devices on the 16" and the latest 13" MBP models. Apple have fixed this in 10.15.6 that came out a few days ago.

per1234, 2020-08-01:
@paris007 does this issue still occur for you when using macOX 10.15.6?

codingcoast, 2020-08-11:
I have this issue with OSC 10.15.6
Arduino IDE is 1.8.13

When I go to System diagnose -> Hardware -> USB I can see the ATm32U4DFU connected but Serial Port in the IDE doesn't show up.

DGEllis, 2020-09-23:
I have a late 2018 Mac Mini running Catalina 10.15.6. I have two Arduino MKR WAN 1310s. One I can connect to a USB 2 port and can use. The other I cannot get to connect in any way. I have tried both USB2 and USB3. I have also tried Catalina 10.15.6, Windows 10 and Fedora 32 Linux on a dual boot PC.

One works, the other doesn't.

closed 2020-09-23 by per1234