arduino/Arduino #10511: Arduino IDE on Linux incorrectly executes a sketch folder

MaksimBalas, 2020-07-14:
Hi developers! Thank you for this magical product!

If I install the Arduino IDE in the latest Ubuntu Studio via the Snap Store, everything works fine. If I install the Arduino IDE on Xubuntu 18.04.4 by downloading arduino-1.8.13-linux32.tar.gz and unpacking it into the arduino-1.8.13 folder (then ./install.sh from root and ./arduino-linux-setup.sh ), everything works fine.

If in latest Ubuntu Studio download arduino-1.8.13-linux64.tar.xz, then extract and move to ~/Arduino folder, Arduino IDE starts normally. After closing, the second time does not start, only shows a splash image for a second and does not start. If I erase ~/.arduino15, the IDE starts normally again, but only once. However, it is impossible to find information in the log files that the program did not like. As a result, it turned out that it was necessary to select a folder for sketches in File - Settings that is different from ~/Arduino, then everything starts working fine.

There is a suggestion when such an event occurs, display a modal window with a proposal to select a separate folder for sketches.

Thank you!

matthijskooijman, 2020-07-14:
Nice how people keep finding ways to break the basic assumptions of the developers :-)

The `~/Arduino` folder is the default sketchbook location, and apparently all kinds of things break when the installation is inside (maybe also below) that particular folder. I suspect that some checks might get confused about whether certain files are in your sketchbook or install directory (since they are in both).

Thanks for reporting this, it would indeed seem good to check against this case on startup (I thought we had a similar check in place, but maybe it's slightly different, I'm misremembering or the check is broken). I'm not entirely sure how to handle this properly, though. The easy way would be to show an error message and then quit, requiring the user to move the Arduino installation elsewhere. However, a user might want to change their sketchbook location instead, but that requires more work (you say that the first startup is successful, so we could show a prompt and let the user edit the sketchbook through preferences normally, but I don't think we can assume that this first startup will always be successful, it might break in the future). So that means we should show a prompt that allows changing the sketchbook location as part of that prompt, which is possible, but a bit more work.

Thoughts welcome, if anyone wants to implement this, also welcome (I won't have time, probably).

per1234, 2020-08-01:
>I thought we had a similar check in place, but maybe it's slightly different, I'm misremembering or the check is broken

It was added here: https://github.com/arduino/Arduino/commit/bede6967d5f0eaaeb927f43ac1168912e5dc1c98 and has apparently since broken or been removed.
[image]

>we should show a prompt that allows changing the sketchbook location as part of that prompt, which is possible, but a bit more work.

I think that is the ideal solution. This was previously suggested at https://github.com/arduino/Arduino/issues/4995