arduino/Arduino #10518: [Library manager] Remove 1.0 version of Grandeur Cloud Library

abdullahmahboob, 2020-07-18:
When I requested to add the Grandeur Cloud library to Library Manager, I mistakenly added the wrong version number in library.properties file, even though it was correct on Github. Now I resolved some bugs in the library and released it as 0.1.2 version after syncing the version in library.properties as well. Is there a way I or you guys can remove the previous version aka. 1.0 from the library manager? It is misleading the users because they think 1.0 is the latest version which it is not.

Thanks in advance.

per1234, 2020-07-18:
We can do that, but first you need to delete the [`0.1.0-beta`](https://github.com/grandeurtech/grandeurcloud-arduino-sdk/releases/tag/0.1.0-beta), [`0.1.0`](https://github.com/grandeurtech/grandeurcloud-arduino-sdk/releases/tag/0.1.0), and [`0.1.1`](https://github.com/grandeurtech/grandeurcloud-arduino-sdk/releases/tag/0.1.1) tags (all of which use the incorrect "1.0.0" `version` value in library.properties.

You can do this by going to each of the links in the previous sentence and clicking the "Delete" button.

abdullahmahboob, 2020-07-19:
Deleted all faulty releases.

cmaglie, 2020-07-24:
Done.

closed 2020-07-24 by cmaglie