arduino/Arduino #10519: Arduino MKR RGB Shield

kckemper, 2020-07-18:
The link to the eagle design files leads to a 404 error.  Here is the link that is mapped to the "eagle files in .zip" button:
https://store.arduino.cc/usa/MKRRGBShield-reference.zip

It sends me to:
https://store.arduino.cc/usa/no-route

kckemper, 2020-07-18:
Looks like the link should be:
https://content.arduino.cc/assets/MKRRGBShieldV3.0.zip

kckemper, 2020-07-18:
Also the store page lists it as 27 mm wide but it's 27.5 mm.

per1234, 2022-07-17:
> The link to the eagle design files leads to a 404 error

I see this error has been fixed.

> the store page lists it as 27 mm wide but it's 27.5 mm.

This information is still present:

- https://store.arduino.cc/products/arduino-mkr-rgb-shield#tech-specs
- https://docs.arduino.cc/hardware/mkr-rgb-shield#panel-teach-spec

So we will consider this issue to be exclusively about the dimensions error now.