arduino/Arduino #10520: [Serial Plotter] Request for XY-plotter

johan1937, 2020-07-19:
The serial plotter in the Arduino IDE does a fine job plotting one or several Y-axis values along the millisecond X-axis. I guess it could be a simple thing to do to include the possibility to plot an XY-chart of two values.
Simplest mode: just read two values from one line from the serial input and plot the corresponding pixel. This needs a clear button.
Advanced features: Have the pixels fade in some seconds. Draw a line from pixel to pixel. Draw a Bezier spline from pixel to pixel. Read two or more pairs of values from each line and plot them with different colours, scale the coordinate system dynamically.

This feature has been asked on the arduino.cc forum and in the comments people have suggested transfering the data to Excel to get it drawn, but that's not a realtime solution. You want to twist some knobs on your circuit while looking at a changing hysteresis curve.

RobTillaart, 2020-07-21:
@johan1937 
No experience but this looks quite good - https://www.megunolink.com/documentation/plotting/xy-plot/

johan1937, 2020-07-21:
Yes, I'm using Meguno right now. The problem is that the Arduino IDE needs the same COM port as Meguno to communicate with the Arduino board. And that can't happen at the same time. Maybe some special settings somewhere can allow two programs in Windows use the same COM port with the same external device.

RobTillaart, 2020-07-21:
No experience - https://freeserialportsplitter.com/