arduino/Arduino #10521: After remapping SPI pins to alternate on STM32F103C8 and using I2C (call Wire.begin()) SPI don't work

vitmailru, 2020-07-19:
Example for reprodusing:

`#include <SPI.h>
#include <SD.h>
#include <Wire.h>

// set up variables using the SD utility library functions:
Sd2Card card;
SdVolume volume;
SdFile root;

// change this to match your SD shield or module;
// Arduino Ethernet shield: pin 4
// Adafruit SD shields and modules: pin 10
// Sparkfun SD shield: pin 8
// MKRZero SD: SDCARD_SS_PIN
const int chipSelect = PA15;

void setup() {
  SPI.setMOSI(PB5);
  SPI.setMISO(PB4);
  SPI.setSCLK(PB3);

  Wire.begin();
  
  // Open serial communications and wait for port to open:
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }

  Serial.print("\nInitializing SD card...");

  // we'll use the initialization code from the utility libraries
  // since we're just testing if the card is working!
  if (!card.init(SPI_HALF_SPEED, chipSelect)) {
    Serial.println("initialization failed. Things to check:");
    Serial.println("* is a card inserted?");
    Serial.println("* is your wiring correct?");
    Serial.println("* did you change the chipSelect pin to match your shield or module?");
    while (1);
  } else {
    Serial.println("Wiring is correct and a card is present.");
  }
}

void loop(void) {
}`
If comment Wire.begin() all works fine, but with Wire.begin() SPI devices do not work.

per1234, 2020-07-19:
This issue tracker is only to be used to report bugs or feature requests. This topic is more appropriate for the STM32Duino Forum. I'm sure they'll be able to help you with your problem over there.

https://www.stm32duino.com/

closed 2020-07-19 by per1234

vitmailru, 2020-07-20:
>>This issue tracker is only to be used to report bugs or feature requests
It's a bug in arduino library,  It's not topic for the forum.

per1234, 2020-07-20:
It may be a bug in an Arduino library, but since you're using an STM32F103C8 I know it's not a bug in an official Arduino library. 

There are hundreds of third party boards platforms and thousands of third party libraries. This issue tracker is only for reporting bugs or feature requests for official Arduino software. For support with 3rd party software, you'll have to go elsewhere.