arduino/Arduino #10544: [Library Manager] Please Add the SpinWearables library

Krastanov, 2020-07-23:
Please add https://github.com/SpinWearables/SpinWearablesFirmware to the library manager.

This library supports a small educational kit and multiple CS and Physics activities based around it. Many of the activities are described at https://spinwearables.com/book/

Krastanov, 2020-08-06:
@per1234 , sorry to bother you with this, but I was wondering whether I am doing something wrong as library addition issues that are more recent have already been addressed, but this one has not been added yet. Am I missing something?

per1234, 2020-08-06:
Hi @Krastanov. I did a cursory scan of your library and it looks perfectly fine. So no, you aren't doing anything wrong.

Your library will not be available for installation via Library Manager until it has been added to the index. cmaglie occasionally comes through to clear all the outstanding Library Manager inclusion requests and closes those issues when that is done. You will receive a notification from GitHub of this. There may be a delay of some hours after that for the library's releases to be picked up by the indexer and for the updates to the [Library Manager index](http://downloads.arduino.cc/libraries/library_index.json) to propagate through the CDN cache. So we'll get to it soon, please be patient.

cmaglie, 2020-08-06:
Done, sorry for the long delay.

closed 2020-08-06 by cmaglie

Krastanov, 2020-08-06:
Thank you both, I appreciate the help!