arduino/Arduino #10556: Add to Library Manager

airgradienthq, 2020-07-25:
Please add https://github.com/airgradienthq/arduino/ to the Library Manager.
Thank You!

airgradienthq, 2020-08-04:
Hi, I am not seeing our library in the lib manager yet. How often is it updated or is there a problem with our library?

per1234, 2020-08-04:
Hi @airgradienthq I did a cursory check of your library and didn't spot any problems, so I think everything is fine on your end.

Your library will not be available for installation via Library Manager until it has been added to the index. cmaglie occasionally comes through to clear all the outstanding Library Manager inclusion requests and closes those issues when that is done. You will receive a notification from GitHub of this. There may be a delay of some hours after that for the library's releases to be picked up by the indexer and for the updates to the [Library Manager index](http://downloads.arduino.cc/libraries/library_index.json) to propagate through the CDN cache. So we'll get to it soon, please be patient.

cmaglie, 2020-08-06:
Done, sorry for the long delay.

closed 2020-08-06 by cmaglie