arduino/Arduino #10565: Filename or extension is too long

nicwilson58, 2020-07-28:
I have been compiling Marlin firmware for a long time with the arduino IDE right up until the last version 2.0.5,  but since the new version 2.0.6 I am now getting a filename or extension is too long error.  I have copied the entire last part of the error below.  I have tried shortening every path I can think of,  temp folders,  sketch folders, I added a short build.path and settings.path to the preferences.txt but nothing so far has helped.

Error,  final part of the build where the error occurs linking.

Linking everything together...
"C:\\arduino1\\hardware\\tools\\avr/bin/avr-gcc" -w -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections,--relax -mmcu=atmega2560 -o "c:\\temp/Marlin.ino.elf" "sketch\\Marlin.ino.cpp.o" "sketch\\src\\HAL\\DUE\\usb\\ctrl_access.c.o" "sketch\\src\\HAL\\DUE\\usb\\sysclk.c.o" "sketch\\src\\HAL\\DUE\\usb\\udc.c.o" "sketch\\src\\HAL\\DUE\\usb\\udi_cdc.c.o" "sketch\\src\\HAL\\DUE\\usb\\udi_cdc_desc.c.o" "sketch\\src\\HAL\\DUE\\usb\\udi_composite_desc.c.o" "sketch\\src\\HAL\\DUE\\usb\\udi_msc.c.o" "sketch\\src\\HAL\\DUE\\usb\\uotghs_device_due.c.o" "sketch\\src\\HAL\\DUE\\usb\\usb_task.c.o" "sketch\\src\\HAL\\LPC1768\\include\\digipot_mcp4451_I2C_routines.c.o" "sketch\\src\\HAL\\LPC1768\\include\\i2c_util.c.o" "sketch\\src\\HAL\\LPC1768\\u8g\\LCD_pin_routines.c.o" "sketch\\src\\HAL\\AVR\\HAL.cpp.o" "sketch\\src\\HAL\\AVR\\HAL_SPI.cpp.o" "sketch\\src\\HAL\\AVR\\MarlinSerial.cpp.o" "sketch\\src\\HAL\\AVR\\Servo.cpp.o" "sketch\\src\\HAL\\AVR\\eeprom.cpp.o" "sketch\\src\\HAL\\AVR\\fast_pwm.cpp.o" "sketch\\src\\HAL\\AVR\\fastio.cpp.o" "sketch\\src\\HAL\\AVR\\u8g_com_HAL_AVR_sw_spi.cpp.o" "sketch\\src\\HAL\\AVR\\watchdog.cpp.o" "sketch\\src\\HAL\\DUE\\DebugMonitor.cpp.o" "sketch\\src\\HAL\\DUE\\EepromEmulation.cpp.o" "sketch\\src\\HAL\\DUE\\HAL.cpp.o" "sketch\\src\\HAL\\DUE\\HAL_SPI.cpp.o" "sketch\\src\\HAL\\DUE\\InterruptVectors.cpp.o" "sketch\\src\\HAL\\DUE\\MarlinSerial.cpp.o" "sketch\\src\\HAL\\DUE\\MarlinSerialUSB.cpp.o" "sketch\\src\\HAL\\DUE\\Servo.cpp.o" "sketch\\src\\HAL\\DUE\\Tone.cpp.o" "sketch\\src\\HAL\\DUE\\dogm\\u8g_com_HAL_DUE_shared_hw_spi.cpp.o" "sketch\\src\\HAL\\DUE\\dogm\\u8g_com_HAL_DUE_st7920_sw_spi.cpp.o" "sketch\\src\\HAL\\DUE\\dogm\\u8g_com_HAL_DUE_sw_spi.cpp.o" "sketch\\src\\HAL\\DUE\\dogm\\u8g_com_HAL_DUE_sw_spi_shared.cpp.o" "sketch\\src\\HAL\\DUE\\eeprom.cpp.o" "sketch\\src\\HAL\\DUE\\eeprom_flash.cpp.o" "sketch\\src\\HAL\\DUE\\eeprom_wired.cpp.o" "sketch\\src\\HAL\\DUE\\fastio\\G2_PWM.cpp.o" "sketch\\src\\HAL\\DUE\\timers.cpp.o" "sketch\\src\\HAL\\DUE\\usb\\sd_mmc_spi_mem.cpp.o" "sketch\\src\\HAL\\DUE\\watchdog.cpp.o" "sketch\\src\\HAL\\ESP32\\FlushableHardwareSerial.cpp.o" "sketch\\src\\HAL\\ESP32\\HAL.cpp.o" "sketch\\src\\HAL\\ESP32\\HAL_SPI.cpp.o" "sketch\\src\\HAL\\ESP32\\Servo.cpp.o" "sketch\\src\\HAL\\ESP32\\WebSocketSerial.cpp.o" "sketch\\src\\HAL\\ESP32\\eeprom.cpp.o" "sketch\\src\\HAL\\ESP32\\eeprom_impl.cpp.o" "sketch\\src\\HAL\\ESP32\\i2s.cpp.o" "sketch\\src\\HAL\\ESP32\\ota.cpp.o" "sketch\\src\\HAL\\ESP32\\spiffs.cpp.o" "sketch\\src\\HAL\\ESP32\\timers.cpp.o" "sketch\\src\\HAL\\ESP32\\watchdog.cpp.o" "sketch\\src\\HAL\\ESP32\\web.cpp.o" "sketch\\src\\HAL\\ESP32\\wifi.cpp.o" "sketch\\src\\HAL\\LINUX\\HAL.cpp.o" "sketch\\src\\HAL\\LINUX\\arduino.cpp.o" "sketch\\src\\HAL\\LINUX\\eeprom.cpp.o" "sketch\\src\\HAL\\LINUX\\eeprom_impl.cpp.o" "sketch\\src\\HAL\\LINUX\\hardware\\Clock.cpp.o" "sketch\\src\\HAL\\LINUX\\hardware\\Gpio.cpp.o" "sketch\\src\\HAL\\LINUX\\hardware\\Heater.cpp.o" "sketch\\src\\HAL\\LINUX\\hardware\\IOLoggerCSV.cpp.o" "sketch\\src\\HAL\\LINUX\\hardware\\LinearAxis.cpp.o" "sketch\\src\\HAL\\LINUX\\hardware\\Timer.cpp.o" "sketch\\src\\HAL\\LINUX\\include\\pinmapping.cpp.o" "sketch\\src\\HAL\\LINUX\\main.cpp.o" "sketch\\src\\HAL\\LINUX\\timers.cpp.o" "sketch\\src\\HAL\\LINUX\\watchdog.cpp.o" "sketch\\src\\HAL\\LPC1768\\DebugMonitor.cpp.o" "sketch\\src\\HAL\\LPC1768\\HAL.cpp.o" "sketch\\src\\HAL\\LPC1768\\HAL_SPI.cpp.o" "sketch\\src\\HAL\\LPC1768\\MarlinSerial.cpp.o" "sketch\\src\\HAL\\LPC1768\\eeprom_flash.cpp.o" "sketch\\src\\HAL\\LPC1768\\eeprom_sdcard.cpp.o" "sketch\\src\\HAL\\LPC1768\\eeprom_wired.cpp.o" "sketch\\src\\HAL\\LPC1768\\fast_pwm.cpp.o" "sketch\\src\\HAL\\LPC1768\\main.cpp.o" "sketch\\src\\HAL\\LPC1768\\timers.cpp.o" "sketch\\src\\HAL\\LPC1768\\u8g\\LCD_I2C_routines.cpp.o" "sketch\\src\\HAL\\LPC1768\\u8g\\u8g_com_HAL_LPC1768_hw_spi.cpp.o" "sketch\\src\\HAL\\LPC1768\\u8g\\u8g_com_HAL_LPC1768_ssd_hw_i2c.cpp.o" "sketch\\src\\HAL\\LPC1768\\u8g\\u8g_com_HAL_LPC1768_st7920_hw_spi.cpp.o" "sketch\\src\\HAL\\LPC1768\\u8g\\u8g_com_HAL_LPC1768_st7920_sw_spi.cpp.o" "sketch\\src\\HAL\\LPC1768\\u8g\\u8g_com_HAL_LPC1768_sw_spi.cpp.o" "sketch\\src\\HAL\\LPC1768\\usb_serial.cpp.o" "sketch\\src\\HAL\\LPC1768\\watchdog.cpp.o" "sketch\\src\\HAL\\SAMD51\\HAL.cpp.o" "sketch\\src\\HAL\\SAMD51\\HAL_SPI.cpp.o" "sketch\\src\\HAL\\SAMD51\\MarlinSerial_AGCM4.cpp.o" "sketch\\src\\HAL\\SAMD51\\QSPIFlash.cpp.o" "sketch\\src\\HAL\\SAMD51\\Servo.cpp.o" "sketch\\src\\HAL\\SAMD51\\eeprom.cpp.o" "sketch\\src\\HAL\\SAMD51\\eeprom_flash.cpp.o" "sketch\\src\\HAL\\SAMD51\\eeprom_qspi.cpp.o" "sketch\\src\\HAL\\SAMD51\\eeprom_wired.cpp.o" "sketch\\src\\HAL\\SAMD51\\timers.cpp.o" "sketch\\src\\HAL\\SAMD51\\watchdog.cpp.o" "sketch\\src\\HAL\\STM32F1\\HAL.cpp.o" "sketch\\src\\HAL\\STM32F1\\HAL_SPI.cpp.o" "sketch\\src\\HAL\\STM32F1\\SPI.cpp.o" "sketch\\src\\HAL\\STM32F1\\Servo.cpp.o" "sketch\\src\\HAL\\STM32F1\\SoftwareSerial.cpp.o" "sketch\\src\\HAL\\STM32F1\\dogm\\u8g_com_stm32duino_fsmc.cpp.o" "sketch\\src\\HAL\\STM32F1\\dogm\\u8g_com_stm32duino_swspi.cpp.o" "sketch\\src\\HAL\\STM32F1\\eeprom.cpp.o" "sketch\\src\\HAL\\STM32F1\\eeprom_bl24cxx.cpp.o" "sketch\\src\\HAL\\STM32F1\\eeprom_flash.cpp.o" "sketch\\src\\HAL\\STM32F1\\eeprom_if_iic.cpp.o" "sketch\\src\\HAL\\STM32F1\\eeprom_sdcard.cpp.o" "sketch\\src\\HAL\\STM32F1\\eeprom_wired.cpp.o" "sketch\\src\\HAL\\STM32F1\\msc_sd.cpp.o" "sketch\\src\\HAL\\STM32F1\\onboard_sd.cpp.o" "sketch\\src\\HAL\\STM32F1\\sdio.cpp.o" "sketch\\src\\HAL\\STM32F1\\timers.cpp.o" "sketch\\src\\HAL\\STM32F1\\watchdog.cpp.o" "sketch\\src\\HAL\\STM32\\HAL.cpp.o" "sketch\\src\\HAL\\STM32\\HAL_SPI.cpp.o" "sketch\\src\\HAL\\STM32\\MarlinSerial.cpp.o" "sketch\\src\\HAL\\STM32\\Sd2Card_sdio_stm32duino.cpp.o" "sketch\\src\\HAL\\STM32\\Servo.cpp.o" "sketch\\src\\HAL\\STM32\\SoftwareSerial.cpp.o" "sketch\\src\\HAL\\STM32\\eeprom_flash.cpp.o" "sketch\\src\\HAL\\STM32\\eeprom_impl.cpp.o" "sketch\\src\\HAL\\STM32\\eeprom_sdcard.cpp.o" "sketch\\src\\HAL\\STM32\\eeprom_sram.cpp.o" "sketch\\src\\HAL\\STM32\\eeprom_wired.cpp.o" "sketch\\src\\HAL\\STM32\\fastio.cpp.o" "sketch\\src\\HAL\\STM32\\timers.cpp.o" "sketch\\src\\HAL\\STM32\\usb_serial.cpp.o" "sketch\\src\\HAL\\STM32\\watchdog.cpp.o" "sketch\\src\\HAL\\STM32_F4_F7\\EmulatedEeprom.cpp.o" "sketch\\src\\HAL\\STM32_F4_F7\\HAL.cpp.o" "sketch\\src\\HAL\\STM32_F4_F7\\HAL_SPI.cpp.o" "sketch\\src\\HAL\\STM32_F4_F7\\STM32F4\\timers.cpp.o" "sketch\\src\\HAL\\STM32_F4_F7\\STM32F7\\TMC2660.cpp.o" "sketch\\src\\HAL\\STM32_F4_F7\\STM32F7\\timers.cpp.o" "sketch\\src\\HAL\\STM32_F4_F7\\Servo.cpp.o" "sketch\\src\\HAL\\STM32_F4_F7\\eeprom.cpp.o" "sketch\\src\\HAL\\STM32_F4_F7\\eeprom_emul.cpp.o" "sketch\\src\\HAL\\STM32_F4_F7\\eeprom_flash.cpp.o" "sketch\\src\\HAL\\STM32_F4_F7\\eeprom_wired.cpp.o" "sketch\\src\\HAL\\STM32_F4_F7\\watchdog.cpp.o" "sketch\\src\\HAL\\TEENSY31_32\\HAL.cpp.o" "sketch\\src\\HAL\\TEENSY31_32\\HAL_SPI.cpp.o" "sketch\\src\\HAL\\TEENSY31_32\\Servo.cpp.o" "sketch\\src\\HAL\\TEENSY31_32\\eeprom.cpp.o" "sketch\\src\\HAL\\TEENSY31_32\\eeprom_impl.cpp.o" "sketch\\src\\HAL\\TEENSY31_32\\timers.cpp.o" "sketch\\src\\HAL\\TEENSY31_32\\watchdog.cpp.o" "sketch\\src\\HAL\\TEENSY35_36\\HAL.cpp.o" "sketch\\src\\HAL\\TEENSY35_36\\HAL_SPI.cpp.o" "sketch\\src\\HAL\\TEENSY35_36\\Servo.cpp.o" "sketch\\src\\HAL\\TEENSY35_36\\eeprom.cpp.o" "sketch\\src\\HAL\\TEENSY35_36\\timers.cpp.o" "sketch\\src\\HAL\\TEENSY35_36\\watchdog.cpp.o" "sketch\\src\\HAL\\shared\\HAL_spi_L6470.cpp.o" "sketch\\src\\HAL\\shared\\backtrace\\backtrace.cpp.o" "sketch\\src\\HAL\\shared\\backtrace\\unwarm.cpp.o" "sketch\\src\\HAL\\shared\\backtrace\\unwarm_arm.cpp.o" "sketch\\src\\HAL\\shared\\backtrace\\unwarm_thumb.cpp.o" "sketch\\src\\HAL\\shared\\backtrace\\unwarmbytab.cpp.o" "sketch\\src\\HAL\\shared\\backtrace\\unwarmmem.cpp.o" "sketch\\src\\HAL\\shared\\backtrace\\unwinder.cpp.o" "sketch\\src\\HAL\\shared\\backtrace\\unwmemaccess.cpp.o" "sketch\\src\\HAL\\shared\\eeprom [... cut ...] glib\\clib\\u8g_rot.c.o" "libraries\\U8glib\\clib\\u8g_scale.c.o" "libraries\\U8glib\\clib\\u8g_state.c.o" "libraries\\U8glib\\clib\\u8g_u16toa.c.o" "libraries\\U8glib\\clib\\u8g_u8toa.c.o" "libraries\\U8glib\\clib\\u8g_virtual_screen.c.o" "core\\core.a" "-Lc:\\temp" -lm
Using library U8glib at version 1.19.1 in folder: C:\arduino\libraries\U8glib 
fork/exec C:\arduino1\hardware\tools\avr/bin/avr-gcc.exe: The filename or extension is too long.
Error compiling for board Arduino Mega or Mega 2560.

nicwilson58, 2020-07-29:
I copied the project to my Macbook Pro running 1.8.12 and found it compiles fine,  this issue seems to be only in the windows version.

per1234, 2021-03-28:
Good news, everyone! This bug has now been fixed by https://github.com/arduino/arduino-cli/pull/961. If you encounter this, please use the hourly build of the Arduino IDE, which has the fix (1.8.13 does not). You can download the nightly here:
https://www.arduino.cc/en/software#hourly-builds

closed 2021-03-28 by per1234