arduino/Arduino #10572: Arduino IDE crashes on start.

joshred83, 2020-07-30:
Arduino IDE shows the splashscreen and then immediately disappears. 

Running "Linux Mint"  v. 20, amd64

per1234, 2020-07-30:
hi @joshred83 Which version of the Arduino IDE are you using?

Please open the folder that contains the Arduino IDE installation in the terminal and then run the IDE from there:
```
./arduino
```
The post here the output you get in the terminal.

joshred83, 2020-07-30:
Oh! It at gave me a more informative error message. Maybe I can troubleshoot this. 

```
(base) joshred@Josh-Mint:~/Arduino/arduino-1.8.13$ ./arduino
Picked up JAVA_TOOL_OPTIONS: 
Set log4j store directory /home/joshred/.arduino15
Incorrect IDE installation folder: Your copy of the IDE is installed in a subfolder of your sketchbook.
Please move the IDE to another folder.
```

per1234, 2020-07-30:
`~/Arduino` is the default sketchbook location. Installing the IDE under that location causes this error.

The easiest fix is to simply move the IDE installation to any other path that's not under `~/Arduino`.

The alternative is to configure a different sketchbook location. Normally, you could do this in the Arduino IDE at **File > Preferences > Sketchbook location**, but that's not possible when the IDE won't start. But you can still configure the sketchbook location by setting the custom sketchbook path in the `sketchbook.path` property of the Arduino IDE's configuration file: `~/.arduino15/preferences.txt`. For example, if you wanted your sketchbook to be at `~/ArduinoSketchbook`, you could do this:
```
mkdir ~/ArduinoSketchbook
echo "sketchbook.path=/home/joshred/ArduinoSketchbook" > ~/.arduino15/preferences.txt
```

We already have another issue about this exact problem, with discussion about how we can make the IDE more user friendly in this situation:
https://github.com/arduino/Arduino/issues/10511
So I'll go ahead and close this one.

closed 2020-07-30 by per1234