arduino/Arduino #10579: Latest IDE board manager fails to load SAMX DUE when running on ubuntu 20.0

jbirdlebough, 2020-07-31:
I just built from source, bossa, toolchain, full latest Arduino IDE version running on ubuntu 20.   When I try to install the DUE SAMX with board manager it fails complaing of bossa is missing and bits of toolchain.   It does work on Win10.  What is needed to make it work on ubuntu 20?read the Issues section of the Contributing Rules at the "Contributing" link to the right before submitting an issue report.

per1234, 2020-09-30:
The problem is that @jbirdlebough is using ARM Linux 64 bit, but Arduino didn't provide the `arm-none-eabi-gcc` 4.8.3-2014q1 and `bossac` 1.6.1-arduino tools dependencies of the `arduino:sam` boards platform for that host.

@jbirdlebough also posted about this on the Arduino forum and I replied there with a workaround:
https://forum.arduino.cc/index.php?topic=698339.0

closed 2020-09-30 by per1234

mwimmer88, 2025-01-16:
there is this SAMX DUE Core provided by per1234 for arm64

to install it via ardunio cli by:

Update ~/.arduino15/arduino-cli.yaml to include the new URL:

```
board_manager:
 additional_urls:
   - https://per1234.github.io/ArduinoCore-sam/package_per1234_samarm64_index.json
```

Update the core index:

`arduino-cli core update-index`

Install the Core

Use the following command to install the per1234:sam core, which is the version with ARM64 support:

`arduino-cli core install per1234:sam`