arduino/Arduino #10580: Null Pointer Exception with no AVR Core installed

NicoHood, 2020-08-01:
Also see: https://bugs.archlinux.org/task/67452

The Arduino IDE should handle the case when no avr core is installed. I remember that worked before, so it must be a problem of some newer versions? Since not everybody wants to use the IDE with AVR we should not rely on it. Especially on archlinux I left the choice to the user to either use the upstream arduino avr core or the archlinux avr core.

NicoHood, 2021-02-14:
The issue still exists and archlinux users open bug reports/email me over and over again about this issue. I guess it is quite easy to fix, can you please have a look at this again?

NicoHood, 2021-04-21:
*push* Please are reporting this issue over and over again

cmaglie, 2021-04-21:
> The issue still exists and archlinux users open bug reports/email me over and over again about this issue. I guess it is quite easy to fix, can you please have a look at this again?

I tried a `light_bundle` build and the IDE seems to start just fine:
[image]

did you have a way to reproduce the issue? maybe it's has been fixed after 1.8.13?

NicoHood, 2021-04-22:
We are also using a light bundle:
https://github.com/archlinux/svntogit-community/blob/packages/arduino/trunk/PKGBUILD

It is not fixed in the recent release, as users still reported the issue. You can reproduce this on a fresh archlinux install. The user folder must not have any arduino related content (like the avr boards) and the package itself should also not ship the arduino avr boards.

cmaglie, 2021-04-22:
if I build a light bundle from the 1.8.13 I got this error:

```
     [exec] java.lang.NullPointerException
     [exec] 	at processing.app.Base.rebuildProgrammerMenu(Base.java:1705)
     [exec] 	at processing.app.Base.<init>(Base.java:286)
     [exec] 	at processing.app.Base.main(Base.java:150)
```

If I build from `master` the IDE starts cleanly, so I assume it was fixed in a way or another. Is the error above that are you experiencing?

If not, may you send me a stack trace or something else to debug it? I'm having trouble reproducing it on the `master` branch.

NicoHood, 2021-04-22:
That is the error I get:
```
$ arduino
Picked up JAVA_TOOL_OPTIONS: 
Set log4j store directory /home/user/.arduino15
java.lang.NullPointerException
	at processing.app.Base.rebuildProgrammerMenu(Base.java:1705)
	at processing.app.Base.<init>(Base.java:286)
	at processing.app.Base.main(Base.java:150)
```

Maybe it got fixed, we will see with 1.8.14. When will it be released?

Offtopic: Can you also have a look at https://github.com/arduino/arduino-cli/issues/1265 please?

NicoHood, 2021-05-13:
It was fixed with 1.8.14, thanks a lot!

closed 2021-05-13 by NicoHood