arduino/Arduino #10583: #include esp32-hal.h (in hardwareserial.h) no such file

Sjaako, 2020-08-01:
I recently used arduino IDE to programm an ESP32 succesfully.
After that I had to upload a programm to some Arduino Nano's .
Selecting the AVR nano and the bootloader i get the above error.
whats wrong?
Selecting my esp32 project and compiling works fine.

per1234, 2020-08-01:
When you encounter an error you'll see a button on the right side of the orange bar in the Arduino IDE "Copy error messages". Click that button. [Paste](https://lifehacker.com/how-to-copy-cut-and-paste-for-beginners-5801525) the error in a message here using [code fencing](https://guides.github.com/features/mastering-markdown/).

Sjaako, 2020-08-02:
Hi Per,
Code Fencing ?? i am new round here... sorry

here is the message:
Arduino:1.8.13 (Windows 10), Board:"Arduino Nano, ATmega328P (Old Bootloader)

In file included from C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:233:0,
from C:\Users\hammo\Desktop\Arduino\Carlo_B3-Blue_Analog_Preset_and_Pots\Carlo_B3-Blue_Analog_Preset_and_Pots.ino:43:
C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/HardwareSerial.h:44:10: fatal error: esp32-hal.h: No such file or directory
 #include "esp32-hal.h"

          ^~~~~~~~~~~~~

compilation terminated.

exit status 1

Fout bij het compileren voor board Arduino Nano

Dit rapport zou meer informatie bevatten met
"Uitgebreide uitvoer weergeven tijden compilatie"
optie aan in Bestand -> Voorkeuren.

per1234, 2020-08-02:
It looks like you somehow replaced or corrupted your Arduino AVR Boards platform installation when you installed the ESP32 boards platform.

Since this is caused by a non-standard use of 3rd party software, rather than a bug in official Arduino software, 
this topic is more appropriate for the Arduino Forum. I'm sure we'll be able to help you with your problem over there.

Please do this:
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules. Be sure to post that full error, using code tags (`</>` button on the toolbar).

closed 2020-08-02 by per1234