arduino/Arduino #10587: Using Cascadia Code font shows garbled text

sikkepitje, 2020-08-03:
This is a bug report. 

I used the Arduino IDE with a modified font : Cascadia Code. I have modified `preferences.txt` to include the line: 
`editor.font=Cascadia Code,plain,16`

All was well, until I installed the preview build of Windows Terminal from the Windows Store, and i believe a modified version of the Cascadia Code font was then installed as well. When I started up Arduino IDE after this, some function names were garbled and unrecognizable. I confirmed the following:
1. this is a result of setting 'Cascadia Code' editor font in Arduino IDE. When I set another font, the issue is not present. 
2. The issue is purely visual. When I copied the source with the garbled charactes to i.e. Visual Studio Code, the source code was intact, what is should be.
3. This is in Arduino IDE 1.8.12.  I tested and confirmed that the issue is also present in the latest "hourly build" of Arduino IDE, at this time version 1.8.14 dated 2020/7/3 04:33 which is on the website labeled  "LAST UPDATE 23 July 2020 16:25:17 GMT". However the function names setup and loop were not garbled, but instead the module name Serial was.

Again I *suspect* that it has something to to with installation of Windows Terminal as I happen to know that a newer font "Cascadia Code" was installed along with it. This is Windows Terminal version: 1.1.2021.0. The release notes says: This version of Windows Terminal ships with Cascadia Code 2007.15.

Screen shots:
https://sikkepitje.nl/w/uploads/Main/Arduino_1.8.12.png
https://sikkepitje.nl/w/uploads/Main/Arduino_1.8.14_Hourly_Build_2020_07_23_04_33.png

per1234, 2020-09-28:
Hi @sikkepitje. You could try this workaround:
1. Open the file `arduino.l4j.ini` you will find in the Arduino IDE installation folder with a text editor.
1. Add a new line to the file:
    ```
    -Dawt.useSystemAAFontSettings=on
    ```
1. Save the file.
1. Restart the Arduino IDE if it's running.

References:
- https://github.com/facchinm/Arduino/commit/fb89a670783d8a45db4998d43ea302ac63147793
- https://github.com/arduino/Arduino/pull/9044

Please let us know if it solves your font problem. Thanks!

sikkepitje, 2020-12-04:
I cannot reproduce the issue anymore. It went away without the suggested fix.