arduino/Arduino #10592: [Library Manager] Please add lvgl and lv_examples

kisvegabor, 2020-08-04:
Hi,

Please add these two libraries into the library manager:

Link to the repositories:
- https://github.com/lvgl/lvgl
- https://github.com/lvgl/lv_examples

And to the library.properties files:
- https://github.com/lvgl/lvgl/blob/master/library.properties
- https://github.com/lvgl/lv_examples/blob/master/library.properties

Thank you!

per1234, 2020-08-04:
Hi @kisvegabor.
Although your `lvgl/lvgl` repository is currently compliant with the requirements for addition to the Arduino Library Manager index, it was not at the time of the last tag (missing library.properties file).

So you'll need to either create a new tag or else redo the existing one.

After that is done, please comment here so we'll know we can proceed.

The `lvgl/lv_examples` repository looks fine as is.

kisvegabor, 2020-08-04:
I fixed it by adding a tag again.

cmaglie, 2020-08-06:
Done.

closed 2020-08-06 by cmaglie

Ali-Abdelwdod, 2021-11-01:
What did you mean by moving theses files to the library.properties files:
    https://github.com/lvgl/lvgl/blob/master/library.properties
    https://github.com/lvgl/lv_examples/blob/master/library.properties