arduino/Arduino #10598: [Library Manager] Please add the BBQ10Keyboard library

arturo182, 2020-08-05:
Please add my library BBQ10Keyboard 1.0.0 to the Arduino library manager

https://github.com/arturo182/arduino_bbq10kbd

Thanks!

per1234, 2020-08-05:
Hi @arturo182. You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

arturo182, 2020-08-05:
Thanks for the quick response, sorry for missing the tag+release, created it now: https://github.com/arturo182/arduino_bbq10kbd/releases/tag/1.0.0

cmaglie, 2020-08-06:
Done.

closed 2020-08-06 by cmaglie