arduino/Arduino #10601: Question: it's Normal to Led 13 is always ON with Bare minimum sketch

va4lak, 2020-08-06:
i have one question about Arduino Uno R3, i have an arduino nearly for 3 years ago. when i was trying the CNC firmware i made a mistake of unplugged the motor when the PSU was still connected and i realice that the pin 13 was ON so i decide to upload a bare minimum sketch to see if this fix the issue BUT unfortunally does not work. recently i receive 2 Brand NEW arduino uno R3 to see if that's a common issue so i try to upload the bare minimum sketch and the two of them make the same behavior,  One of them with the LED less turn ON and the one i have in first place with the LED mostly turn ON (probably because my rookie mistake). i start to think this is a normal thing but i supose you guys are the only one can answer that question or known some technical detail.  i apologize for my english but i'm a spanish speaker. many thank in advance and have a nice day.

CsCrazy85, 2020-09-11:
It's normal that it's on. If you see the schematics you'll see why (the opamp is feeding the LED): https://www.arduino.cc/en/uploads/Main/Arduino_Uno_Rev3-schematic.pdf

closed 2020-09-28 by per1234

va4lak, 2020-10-06:
> 
> 
> It's normal that it's on. If you see the schematics you'll see why (the opamp is feeding the LED): https://www.arduino.cc/en/uploads/Main/Arduino_Uno_Rev3-schematic.pdf

thank you very much CsCrazy85 for taking the time to answer my question, I hope you have an excellent day. Best regards