arduino/Arduino #10610: [Library Manager] EnviroDIY DS3231

SRGDamia1, 2020-08-07:
Another library:  https://github.com/EnviroDIY/Sodaq_DS3231
properties file:  https://raw.githubusercontent.com/EnviroDIY/Sodaq_DS3231/master/library.properties

per1234, 2020-08-07:
Hi @SRGDamia1. The library repository contains a .exe file:
https://github.com/EnviroDIY/Sodaq_DS3231/blob/master/examples/PCsync/PCsync.exe
Libraries in the Arduino Library Manager index are prohibited from containing .exe files:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#how-is-the-library-list-generated

So if you want to have the library added to the Arduino Library Manager index, you'll need to remove the .exe file from the repository and then either create a new tag/release or else redo the existing one.

After that, please comment here to let us know we can proceed. If, instead, you no longer wish to add the library to the Arduino Library Manager, you can go ahead and close the issue.

Thanks!

SRGDamia1, 2020-08-07:
Is there no way to specify which folders or files from the repository to include - something akin to specifying folders for a sparse checkout?

per1234, 2020-08-08:
The Arduino Library Manager uses the Git tag. Whatever is in the tag must be compliant with the requirements specified in the FAQ.

You can move the .exe file to a different branch, add it as a GitHub release asset, or move it to a different repository, but it can't be anywhere in the repository at the ref the tags point to.

SRGDamia1, 2020-08-08:
OK, I'll get it moved.

cmaglie, 2020-08-20:
I've added the library, but until the `exe` is removed the library will be rejected:

```
JOB 3278 - 2020/08/20 16:18:51 Release Sodaq_DS3231:1.3.0 has wrong library name, should be EnviroDIY_DS3231
JOB 3278 - 2020/08/20 16:18:51 Checking out tag: v1.3.2
JOB 3278 - 2020/08/20 16:18:51 Error syncing library: *.exe files found, skipping
```

from now on your repo is automatically checked every hour: just create a release without the .exe and it will be automatically published.

closed 2020-08-20 by cmaglie