arduino/Arduino #10611: [Library Manager] Please add ModularSensors

SRGDamia1, 2020-08-07:
Last one (for today)!

Please add https://github.com/EnviroDIY/ModularSensors

The properties file:  https://github.com/EnviroDIY/ModularSensors/blob/master/library.properties

NOTE:  This one has a lot of dependencies.  I'm not sure how the library manager handles recursive installation of dependencies.  In this case, I *don't* want it to do that because I want to specify the version of the sub-dependencies even when the first level dependency doesn't specify.  I've added the --no-deps flag within the dependency list where that applies.  If the library script just splits at the commas and then runs the lib install command with the text between the commas, it should work very nicely.

cmaglie, 2020-08-20:
> NOTE: This one has a lot of dependencies. I'm not sure how the library manager handles recursive installation of dependencies. In this case, I don't want it to do that because I want to specify the [...]

Well, it **does** recursive install of deps... we can't do what you need here the `depends=..` line you wrote is not supported as is. For now, unless your library can work with the latest of each dep, we do not support your use case yet.

per1234, 2020-09-25:
@SRGDamia1 since your library is still not compliant with the requirements for addition to the Arduino Library Manager index, I'm going to go ahead and close this.

If you ever do resolve the issue of an unsupported library.properties `depends` field format (either  by making it compliant with the [Arduino Library Specification](https://arduino.github.io/arduino-cli/latest/library-specification/#libraryproperties-file-format) or by removing the field from library.properties altogether), and then either make a new tag/release or else redo the existing release, please comment here and I'll be happy to reopen this so we can get your library in the index.

closed 2020-09-25 by per1234