arduino/Arduino #10615: Please stop the library manager from updating the list of libraries EVERY time

thearduinoguy, 2020-08-09:
Please stop the library manager from updating the list of libraries EVERY time.

If I install a library then go back to install another library a few mins later there is n o need for the IDE to yet again check the library list. Nothing will have changed in such a short amount of time. Have an option so it checks only once per day or something. 

This is very annoying when installing several libraries in a short space of time.

per1234, 2020-08-09:
Ever since https://github.com/arduino/Arduino/pull/9023, the Library Manager index has been cached. So, as long as you're using Arduino IDE 1.8.10 or newer, the index will only be downloaded the first time you open Library Manager. The delay you see on subsequent openings of Library Manager is purely the time required to parse the >13 MB Library Manager index file that's already stored on your computer.

thearduinoguy, 2020-08-10:
Why does it say "Updating list of installed libraries" every time I open it? Version 1.8.13

per1234, 2020-08-10:
That's when it's scanning the installed libraries and parsing the index's 13 MB of JSON.

When it's downloading the Library Manager index from the Internet the status bar says "Downloading libraries index".

thearduinoguy, 2020-08-10:
For only a tiny 13Mb file it seems to take a very long time to parse it. Why can it not keep such a small amount of data in memory. Why does it have to keep on parsing it over and over?