arduino/Arduino #10616: [Library Manager] Please add GAMMA library

RobTillaart, 2020-08-09:
Please add https://github.com/RobTillaart/GAMMA

Thanks,
Rob

per1234, 2020-08-09:
Hi @RobTillaart. The `name` value in your library.properties file contains a disallowed character `,`:
https://github.com/RobTillaart/GAMMA/blob/0.1.0/library.properties#L1
```
name=gamma,h
```
From https://arduino.github.io/arduino-cli/latest/library-specification/#libraryproperties-file-format:
>- name - the name of the library. Library names must contain only basic letters (A-Z or a-z) and numbers (0-9), spaces (), underscores (_), dots (.) and dashes (-). It cannot start or end with a space, and also it cannot start with a number. Note that libraries with a name value starting with Arduino will no longer be allowed addition to the Library Manager index as these names are now reserved for official Arduino libraries.

Please correct the `name` value and then either create a new tag/release or else redo the existing one. Once that's done, please comment here so we'll know we can proceed.

Thanks!

RobTillaart, 2020-08-10:
Thanks for noticing,
fixed it,
Rob

cmaglie, 2020-08-20:
Done.

closed 2020-08-20 by cmaglie