arduino/Arduino #10618: [Library Manager] Please add PCA9685-Arduino

NachtRaveVL, 2020-08-10:
Homepage: https://github.com/NachtRaveVL/PCA9685-Arduino

Library file: https://github.com/NachtRaveVL/PCA9685-Arduino/blob/master/library.properties

Thank you!

per1234, 2020-08-10:
Hi @NachtRaveVL. You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

NachtRaveVL, 2020-08-16:
Sorry for the delay on this, try now. :)

cmaglie, 2020-08-20:
Done.