arduino/Arduino #10621: New Release for Library: ModernPIDControlSS

somefunAgba, 2020-08-10:
Library at https://github.com/somefunAgba/ModernPIDControlSS updated to v1.1

per1234, 2020-08-10:
Hi @somefunAgba. The Library Manager indexer checks all the libraries on the list for new tags/releases hourly. After that, it may take some time for the updated index to propagate through the CDN cache to make your new release available everywhere, but it will get there in good time.

So you don't need to create an issue report every time you release a new version; once you push a tag/release that is compliant with all the Library Manager requirements everything happens automagically from there.

Reference:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#how-can-i-publish-a-new-release-once-my-library-is-in-the-list

closed 2020-08-10 by per1234

somefunAgba, 2020-08-10:
thanks. wanted to confirm.