arduino/Arduino #10646: Sad about seeing the SPIFFS library from the lib-manager been dropped...

rin67630, 2020-08-15:
Apparently, SPIFFS was dropped from the official library.

OK, LittleFS should be used instead, but it is NOT a replacement for me, since LittleFS has a minimum block size of 4K, which would mean a dramatic waste of the limited EEPROM space and invoke too much write operations for simple ongoing message logging with messages about 60 bytes length each.
The 256 bytes from SPIFFS are just tremendously more efficient and save unnecessary writes on an EEPROM that is very limited. That block size was already too big for a small EEPROM size, but to impose 4K is just unacceptable!

I can surely get the SPIFFS lib from GitHub and include it manually, but it will unnecessarily complicate the deployment of my project to other users.

Why did you drop a successful working library, that thousands people use?
Is might not be maintained any more, but it worked and had defintely advantages over littleFS.

Please restore the lib as is was, you may include a deprecation warning, but please, let it in the Lib manager.

matthijskooijman, 2020-08-15:
I'm under the impression that SPIFFS is actually a library that is included with the ESP8266 (and maybe also ESP32) cores. Are you sure it used to be installed through the lib manager? If so, this is really the wrong place to report this, and this is something decided by the maintainers of the ESP cores (which is not Arduino itself, they are third-party cores).

If this was a library previously distributed through the library manager, even then it would probably have been dropped on request by the library maintainers (which is again, I think, not Arduino itself). I checked my github notification history, I don't see any request to add or remove the library in the past, so I think this is indeed something internal to the ESP cores...

rin67630, 2020-08-16:
Ah OK, sorry for the mistake. I will forward my request to the right place.

closed 2020-08-16 by rin67630