arduino/Arduino #10647: Bug Report: Library manager's dependencies installation window unusable

jisforjt, 2020-08-15:
[image]

When you go to _Manage Libraries..._ and try to install a library that requires a lot of dependencies, e.g. _Adafruit's Sensor Lab_. A message window appears entitled _Dependencies for library Adafruit Sensor Lab:0.4.2_ with a message that states _The library __Adafruit Sensor Lab:0.4.2__ needs some other library dependencies currently not installed:_ and then lists the libraries you need to install. Normally you would just click _Install all_, _Install [blank] only_, or _Cancel_, but the problem is that the window is so big I can't see or click anything on the bottom. I can't even tab return to install it. A simple slide bar would fix this.

Arduino IDE: v1.8.13
OS: macOS Catalina: v10.15.6
Screen resolution: 1440 x 900

mbncr, 2022-06-15:
Hello. 
I still encounter this issue on Arduino IDE 1.8.19 / Monterey 12.3
Same as @jisforjt , except I can't tab my way out of the Dependencies window.

Tinyu-Zhao, 2022-07-28:
@cmaglie Why has this problem not been solved after two years?
😖

708yamaguchi, 2022-08-22:
I am facing the same problem.
I hope this problem will be solved.