arduino/Arduino #10654: [Library Manager] add wpi-32u4-library to the library manager

madhephaestus, 2020-08-17:
Please read the Issues section of the Contributing Rules at the "Contributing" link to the right before submitting an issue report.
This library provides the support classes for the WPI rbe 2001 course materials. It is a modification of the Pololu library, with additional motor support. 

https://github.com/WPIRoboticsEngineering/wpi-32u4-library

cmaglie, 2020-08-20:
A library called `Romi32U4` is already registered (I guess it's the library that you forked).
You must change the name of the library if you want it added:

1. edit your `library.properties` and change the `name`
2. commit & push
3. delete and redo the latest git tag

madhephaestus, 2020-08-20:
i made the latest release renamed to establish the fork. THank you!

cmaglie, 2020-08-20:
One last change:

https://github.com/WPIRoboticsEngineering/wpi-32u4-library/blob/2.0.1/library.properties#L2

you wrote `version=2.0.` instead of `2.0.1` :-)

madhephaestus, 2020-08-20:
oopsie, fixed. Thanks!

cmaglie, 2020-08-20:
Ok, done.

closed 2020-08-20 by cmaglie