arduino/Arduino #10656: Arduino Due schematic is incorrect w.r.t. native USB port

aaronmdjones, 2020-08-18:
Page 6 of [1] states that the pinout for the native USB port (top right) is the following:

| USB Port Pin | MCU Pin / Net |
| --- | --- |
| GND | GND |
| USB ID | DHSDM |
| USB D+ | DHSDP |
| USB D- | PA25 |
| VBUS | +5V |

However, it should actually be:

| USB Port Pin | MCU Pin / Net |
| --- | --- |
| GND | GND |
| USB ID | PB10 |
| USB D+ | DHSDP |
| USB D- | DHSDM |
| VBUS | +5V |

That is; the assignments for `USB ID` and `USB D-` are the wrong way around, and `UOTGVBOF` (`USB ID`) is MCU pin `PB10`, not `PA25`.

[1] https://content.arduino.cc/assets/Pinout-Due_latest.pdf

aaronmdjones, 2020-08-18:
I have also confirmed this with a PCB trace continuity test; the second-from-the-left pin on the USB port (`UOTGID`) is definitely connected to MCU pin 129 (`PB11`). This is described on page 1057 of [2].

[2] http://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-11057-32-bit-Cortex-M3-Microcontroller-SAM3X-SAM3A_Datasheet.pdf

FunzioneFabio, 2020-09-04:
Hello @aaronmdjones ! Thank you so much for reporting this! The pinout is now corrected and uploaded on the website.
https://content.arduino.cc/assets/Pinout-Due_latest.pdf

closed 2020-09-04 by kengdahl