arduino/Arduino #10662: Debounce Tutorial: Reliability Issues

TimMathias, 2020-08-19:
It's a great tutorial, but there are a couple of issues with obtaining an accurate time stamp for the button sample.

Consider this code snippet from the Debounce tutorial:
```c++
  // read the state of the switch into a local variable:
  int reading = digitalRead(buttonPin);
.
.
.
    lastDebounceTime = millis();
.
.
.
  if ((millis() - lastDebounceTime) > debounceDelay) {
```
Firstly, the vertical ellipses represent lines of code that could be interrupted. The time spent in the associated ISR could throw off the timing of the time stamp.

Secondly, the `millis()` function is called twice,, so it's not using the original time stamp, leading to a potential miscalculation.

IMHO, the Debounce tutorial should highlight these issues.

I've written a potential fix which uses the `ATOMIC_BLOCK` macro to prevent interrupts from inteferring with the time stamp, and it reads the time stamp only once into a local variable for use in subsequent calculations:
```c++
    bool input_state;
    unsigned long current_ms;

    // Temporarily disable interrupts to ensure an accurate time stamp for the sample.
    ATOMIC_BLOCK(ATOMIC_RESTORESTATE)
    {
      input_state = digitalRead(INPUT_PIN);  // No interrupts will occur between here...
      current_ms = millis();                 // ...and here, ensuring an accurate time stamp for the sample.
    }
```
However, I don't know how well supported is the `ATOMIC_BLOCK` macro: across the various Arduino systems.

Here is an example of the potential fix. With this particular implementation of hysteresis, the debouncer doesn't need to know whether the input is active high or active low; it just simply debounces either edge as it meets it:
```c++
/*
  Debouncer

  Debounces an input using hysteresis.
  The LED shows the state of the button.

*/

#include <util/atomic.h>

class Debouncer
{
private:
  const int INPUT_PIN;
  bool output_state;
  bool rise;
  bool fall;
  const unsigned long DEBOUNCE_DELAY_ms;
  unsigned long last_ms;

public:

  Debouncer(const int INPUT_PIN, const unsigned long DEBOUNCE_DELAY_ms = 50)
      :
      INPUT_PIN(INPUT_PIN),
      DEBOUNCE_DELAY_ms(DEBOUNCE_DELAY_ms)
  {
    this->output_state = false;
  }

  bool Output() const { return output_state; }
  bool Rise() const { return rise; }
  bool Fall() const { return fall; }

  bool IsDebounced()
  {
    bool input_state;
    unsigned long current_ms;

    // Temporarily disable interrupts to ensure an accurate time stamp for the sample.
    ATOMIC_BLOCK(ATOMIC_RESTORESTATE)
    {
      input_state = digitalRead(INPUT_PIN);  // No interrupts will occur between here...
      current_ms = millis();                 // ...and here, ensuring an accurate time stamp for the sample.
    }

    rise = fall = false;

    // Hysteresis:
    //   If there is no change, reset the debounce timer.
    //   Else, compare the time difference with the debounce delay.
    if (input_state == output_state)
    {
      last_ms = current_ms;
    }
    else if ((current_ms - last_ms) >= DEBOUNCE_DELAY_ms)
    {
      // Successfully debounced, so update the outputs.
      rise = input_state && !output_state;
      fall = !input_state && output_state;
      output_state = input_state;
      return true;
    }
    return false;
  }
};

const int BUTTON_PIN = 2;  // Pushbutton pin number.
const int LED_PIN = 13;    // LED pin number.
const unsigned long DEBOUNCE_DELAY_ms = 50;  // The debounce time in milliseconds.

void setup()
{
  // Set pin modes.
  pinMode(BUTTON_PIN, INPUT);
  pinMode(LED_PIN, OUTPUT);
}

void loop()
{
  static Debouncer button(BUTTON_PIN, DEBOUNCE_DELAY_ms);
  static bool led_state = false;

  if (button.IsDebounced())
  {
    // Update the LED to show the debounced signal.
    digitalWrite(LED_PIN, button.Output());

    // Alternative feedback: Toggle the LED on the rising edge of the debounced signal.
    /**
    if (button.Rise())
    {
      led_state = !led_state;
      digitalWrite(LED_PIN, led_state);
    }
    /**/
  }
}
```

kengdahl, 2020-08-20:
If I understood the point of the debounce tutorial on the Arduino webpage is give an simple example of how to debounce input signal for a beginner. It is true that it does not show the best way of doing this. But it keeps it simple without introducing a lot of high level concepts.
The '''ATOMIC_BLOCK''' only works on AVR based boards and not the ARM ones. 

Maybe the best solution would be to add this as a tutorial on Project hub?

alranel, 2020-08-31:
Or maybe could you turn this into a library @TimMathias? That one could be linked from the documentation :)

kengdahl, 2020-09-07:
I'm reopening this because we have yet to try this change on all different architectures.

closed 2020-09-07 by kengdahl

TimMathias, 2020-09-09:
@alranel, I've written a [simple debouncer](https://github.com/TimMathias/Arduino-Debouncer/blob/master/debouncer.h). Do you think this is suitable for linking to from the official documentation?

At the moment, it only implements the ATOMIC_BLOCK supported by AVR.

I've asked for advice [here ](https://arduino.stackexchange.com/questions/77494/which-arduinos-support-atomic-block) about implementing it for all Arduino architectures. Ideally, I would like a unique `#define` for each architecture, and accompanying code to save/restore the global interrupt flag for each architecture. And then combine it all into one super macro and/or C++ class.