arduino/Arduino #10663: [Library Manager] update DST_RTC library

andydoro, 2020-08-19:
Previous library version contained errors and should be updated. https://github.com/andydoro/DST_RTC

per1234, 2020-08-20:
Hi @andydoro. Just so I'm sure I understand: did you redo your [`v1.1.0` tag](https://github.com/andydoro/DST_RTC/tree/v1.1.0), and you're now requesting that the outdated 1.1.0 version of the library that's in the Arduino Library Manager index be removed and replaced by the current 1.1.0?

andydoro, 2020-08-20:
That's correct. Sorry if the original message was unclear.

cmaglie, 2020-08-20:
@andydoro 

I can do it but it's not the correct way beacuse:
- users that have already installed 1.1.0 will not be notified of the change
- users will need to uninstall the old-1.1.0 and install the new-1.1.0, but to they can't tell the difference because both old and new will be presented as "1.1.0"

much better if you release an 1.1.1 version with the fix:
- user already having the lib will be asked to upgrade automatically
- new users will install directly the latest 1.1.1

andydoro, 2020-08-20:
Thank you! I've created a v1.1.1. Please let me know if there's anything else I should do.

cmaglie, 2020-08-20:
> Please let me know if there's anything else I should do.

nope, the library is automatically published withing an hour once you create the tag (and it has been published already)

closed 2020-08-20 by cmaglie