arduino/Arduino #10665: Mechanical Documentation

grossman83, 2020-08-20:
I'm aware that this is more for software bugs than it is for a mechanical request, but I've searched high and low for some mechanical documentation and been unable to find it. I love Arduino to the point that I'm putting some in products of mine at my company and therein is the problem. I want to bolt them down, and there is really poor documentation on the mechanical layout of the case that my arduino due came with. I've done what I can to measure, but it would be really nice if in some standard documentation the physical dimensions of the various arduinos was specified.

kengdahl, 2020-08-20:
Would a DXF of the case solve the issue?

grossman83, 2020-08-20:
Yes, a DXF would solve the issue for me. I think a .pdf would help many more than just me and ought to be included in all documentation.

I find it strange that on the Arduino website it is impossible to find the mechanical layout of the Due. Under the documentation tab for the Due I see:
Eagle Files in Zip
Schematic in .pdf
Fritzing in FZPZ

However on the Uno documentation tab I see:
Eagle Files in Zip
Schematic in .pdf
Board Size in .dxf

It is really nice when companies have consistently broad and consistently named documentation across their product lines. Thus, it would be my recommendation that if you are going to supply .dxf of the Uno it also be supplied in the same relative location for the Due and all other Arduino devices. This recommendation also applies to the case and its dimensions. And finally it applies to the Fritzing in FZPZ.

per1234, 2020-09-28:
Related issue: https://github.com/arduino/Arduino/issues/10586

kengdahl, 2020-10-08:
I have added a technical drawing in PDF of the Arduino DUE

closed 2020-10-08 by kengdahl

grossman83, 2020-10-08:
That's really nice. Thank you very much. Not a big deal, but there are some
"internal" holes that are not shown in your drawing and it would be nice to
have those too.

-Marc

kengdahl, 2020-10-22:
@grossman83 I'll see what I can do about it.

kengdahl, 2021-02-02:
@grossman83 I have updated the technical drawing with the internal holes.

grossman83, 2021-02-03:
Awesome!

-Marc

kengdahl, 2021-02-03:
@grossman83 Was there any other products that was missing technical drawings?

grossman83, 2021-02-03:
It's just that one that I'm using so I'm afraid I can't speak for all
arduinos.

-Marc