arduino/Arduino #10694: [Library Manager] please add UCloudIoTSDK to library manager

jamesLiu-ucloud, 2020-08-26:
Please add UCloudIoTSDKto the library manager. An SDK which can help users connect to UCloud iot core platform.

https://github.com/ucloud/ucloud-iot-sdk-arduino

per1234, 2020-08-26:
Hi @jamesLiu-ucloud. You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

jamesLiu-ucloud, 2020-08-27:
> 发送日期：2020-08-27 00:49:38
> 抄送人：jamesLiu-ucloud <james.liu@ucloud.cn>,Mention <mention@noreply.github.com>

Hi.I must to say thank you first! Now I have created a release and the version value is same as value in
library.properties.
I hope you can check it and add it to the library.If there is anything else wrong with my operation ,please tell me with repling this mail. thank you!
address:https://github.com/ucloud/ucloud-iot-sdk-arduino/

 https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

cmaglie, 2020-08-31:
Done.

closed 2020-08-31 by cmaglie