arduino/Arduino #10695: HT1382 library for Arduino

aavestan, 2020-08-26:
Please add HT1382 to the library manager. An SDK which can work with Holtek RTC ic HT1382 .

https://github.com/aavestan/HT1382

cmaglie, 2020-08-31:
You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

per1234, 2020-09-25:
Closing due to lack of feedback. @aavestan if you ever get around to making a release/tag in your library, please comment here and I'll be happy to reopen this so you can get your library in the Arduino Library Manager index.

closed 2020-09-25 by per1234